Intermodalics/tango_ros #127: Allow to run roscore inside the app

PerrineAguiar, 2017-01-25:


PerrineAguiar, 2017-01-25:
@adamantivm Do you have an example on how to this?

jubeira, 2017-01-25:
@PerrineAguiar the default MasterChooser has an option to do this. Look for the method `onActivityResult` inside `RosActivity`, there is a codepath that starts a new master there. Hope it helps!

PerrineAguiar, 2017-01-26:
@jubeira Thanks for pointing me in the right direction.

adamantivm, 2017-01-26:
Sorry I'm a bit swamped travelling at the moment. Thank you @jubeira for the pointer to @PerrineAguiar.

BTW, at some point it would be nice to generalize this "alternate master chooser" model so that new rosjava apps can get this type of behavior by default or as an alternative to the master chooser.

PR #132 (2017-01-27): Feature/run roscore in app
This PR allows to run the ROS master on the device.

Fixes #127.
closed 2017-01-27 by PerrineAguiar