Intermodalics/tango_ros #143: Running tango_ros on Lenovo Phab2

aseyfi, 2017-02-03:
Hello,

Is the app available on Google Play (version 1.0-5) tested with the Lenovo Phab2 phone? I can run the app and see the published topics in my Linux machine, but I cannot see any published data using rostopic echo command.

Thank You,
Ahmad

jubeira, 2017-02-06:
Hello @aseyfi , were you able to see the published data? Did you encounter some problem?

aseyfi, 2017-02-08:
Hi @jubeira, yes, I was able to see the published data running the app on Lenovo Phab2. Somehow the topics in app's menu were getting turned off.

The other issue that I am experiencing is that when I look at the video from fisheye camera in RVIZ, sometimes I see that video freezes for a few seconds. Is it the app's issue or you think there is something wrong with my network?

Thank You,
Ahmad

smits, 2017-02-08:
We haven't seen that yet on our side, except when wifi is really bad. How did you setup your networking, could you retry with USB tethering?

aseyfi, 2017-02-23:
@smits , Thank you for the suggestion. I used USB tethering and now I don't see the issue any more.

Thank You,

aseyfi, 2017-03-02:
@smits , with tethering freezing the frames are not happening, but I am getting low frame rates. Would you please let me know what is the best rate that I can expect from different topics, specifically, color camera, fisheye and depth?

Thank You,
Ahmad

smits, 2017-03-08:
Which version are you using? What framerates are we talking about?

aseyfi, 2017-03-08:
@smits , I am compiling the app based on tag v1.0-13 and run it on Lenovo Phab2 and my question is about the following topics, assuming that I subscribe for all three at the same time:

/tango/camera/color_1/image_rect/compressed
/tango/camera/fisheye_1/image_rect/compressed
/tango/point_cloud

Thank You,
Ahmad

PerrineAguiar, 2017-03-13:
Here are the rates we observed for the following topics:

/tango/camera/color_1/image_rect/compressed ~3 Hz
/tango/camera/fisheye_1/image_rect/compressed ~5 Hz
/tango/point_cloud ~5 Hz

For better rates, I would recommend using the non rectified images if possible :
/tango/camera/color_1/image_raw/compressed ~8 Hz
/tango/camera/fisheye_1/image_raw/compressed ~30 Hz

adamantivm, 2017-03-13:
I'm very curious if subscribed from another node within the same device if the rates are better.
If that is true then we would know that the bottleneck is in the network but also makes it more interesting to add more logic on the device side.

aseyfi, 2017-03-13:
Thank you @PerrineAguiar .@adamantivm, when it comes to publishing compressed color images, [this line](https://github.com/Intermodalics/tango_ros/blob/master/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp#L803) is the bottleneck and its execution takes in average about 110ms on Lenovo Phab2. I have done some experiments to solve these problem by running multiple threads to compress and publish color images [here](https://github.com/aseyfi/tango_ros/blob/systrace/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp#L926) and assigning task to each thread [here](https://github.com/aseyfi/tango_ros/blob/systrace/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp#L783). By running 4 extra threads I was able to publish compressed color images @ ~15Hz. The problem is that since tag v1.0-13, [CameraPublisher](https://github.com/Intermodalics/tango_ros/blob/master/tango_ros_common/tango_ros_native/include/tango_ros_native/tango_ros_node.h#L190) was used for transferring images and it makes it difficult to implement a similar multithreading scheme.

Also I have opened [this issue](https://github.com/Intermodalics/tango_ros/issues/167) for running android native tracing. If we can get it to work, then it would be easier to analyze the code.