Intermodalics/tango_ros #160: Failed to build App from source

smits, 2017-02-22:
I'm testing the #152 but I get the following error, which makes me think the build instructions are not complete. 
```
Errors     << tango_ros_streamer:cmake /home/rsmits/tango_ros_ws/logs/tango_ros_streamer/build.cmake.003.log                      
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "rosjava_build_tools" with any of the following names:

    rosjava_build_toolsConfig.cmake
    rosjava_build_tools-config.cmake
```

I resolved this by running rosdep install --from-path src --ignore-src --rosdistro kinetic -r

On kinetic rosdep install complains about the following missing packages:
```
tango_ros_native: Cannot locate rosdep definition for [opencv2]
glog_catkin: Cannot locate rosdep definition for [gflags_catkin]
```

PerrineAguiar, 2017-03-14:
Ithink this is solved with the new doc. See #176.

smits, 2017-03-21:
Yes you are right. Marking as closed.

closed 2017-03-21 by smits