Intermodalics/tango_ros #165: Consider alternative build instructions using pre-built roscpp_android

adamantivm, 2017-02-28:
It may be possible to skip the build step for roscpp_android and use pre-built binaries instead (see point 2 here: http://wiki.ros.org/android_ndk/Tutorials/Building%20The%20Example%20Applications%20using%20the%20Binary%20Distribution)

Was this attempted?

jubeira, 2017-03-01:
@adamantivm perhaps a new release of [roscpp_android](https://github.com/ekumenlabs/roscpp_android) should be performed for that; that link points to an old release, which doesn't have all the required packages for this app (in particular, [Image transport plugins](https://github.com/ekumenlabs/roscpp_android/pull/56).

adamantivm, 2017-03-01:
Ok, I left an issue in the repo to request a new release - https://github.com/ekumenlabs/roscpp_android/issues/59 . After than then this can be improved to not require the lengthy rosndk build.