Intermodalics/tango_ros #167: Android Native Tracing

aseyfi, 2017-03-02:
Hello,

I am trying to use android tracing in the app, so based on the instruction [here](https://developer.android.com/ndk/guides/tracing.html#overview) I added '#include <android/trace.h>' to [tango_ros_node.cpp](https://github.com/Intermodalics/tango_ros/blob/master/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp) file (I didn't make any other changes), but when I make the project, I am getting this error: 
`fatal error: android/trace.h: No such file or directory`
Would somebody please help me to fix this issue?

Thank you in advance,
Ahmad

smits, 2017-03-13:
When you make the project, you use catkin_make or android studio?  The tracing functionality is only available for API levels 23 and higher (as stated in the link you provided), did you adapt the build files to use this API level?

aseyfi, 2017-03-13:
@smits , I am using Android Studio and API 25.

PerrineAguiar, 2017-03-13:
@aseyfi I think you need to change [this line](https://github.com/Intermodalics/tango_ros/blob/master/TangoRosStreamer/tango_ros_node/src/main/jni/Application.mk#L2) to:
`APP_PLATFORM := android-23`

Also make sure your android ndk is up to date, [here](https://developer.android.com/ndk/downloads/index.html) you can download the latest version. Note that you'll need to update your local.properties consequently.

mekya, 2017-11-08:
for my case, setting minSdkVersion 23 in build.gradle works for me. 

It seems that build.gradle's minSdkVersion is used in compiling native code.