Intermodalics/tango_ros #170: rosbag record on device?

gitunit, 2017-03-09:
latency is very huge when topics are streamed via wifi to a desktop machine.
is there a way to record all messages on device? if not, is this on your roadmap?

also what is the local master for?

thx in advance

smits, 2017-03-13:
The local master can be used to have multiple nodes running on the device (such as an additional rosbag record node ;) ).  It can also be used when using the Tango device as master in a robotic application.

gitunit, 2017-03-16:
thx @smits . do i need to rebuild the project to run the rosbag record node? or is there a simpler way?

smits, 2017-03-17:
It won't work out of the box.  You'll need to figure out how to launch the rosbag node from the ROSJava application or how to get the rosbag executable on the device and launch it through adb.  All of this is not directly on our roadmap.