Intermodalics/tango_ros #184: Look into delays for images after running a while.

PerrineAguiar, 2017-03-16:


PerrineAguiar, 2017-03-31:
Action to take now:
* Run existing unit test over a weekend
* Calculate rate over a window and not all images
* Print rates

Next time this issue happens:
1.  What is the message rate?
2. By how much the message is delayed (communication/serialization overhead)? Use `rostopic echo --offset <topic_name>/header/stamp`
3. Is it the first time the app was stared after a reboot?
4. Does closing rviz solved the issue?
5. Does rebooting the app solved the issue?

mcopejans, 2017-04-06:
The issue we encountered when the app does not start publishing is very likely related to a dynamic reconfigure bug. When toggling the button the parameter_node prints the following:
```
No such service /tango/set_parameters of type dynamic_reconfigure/Reconfigure
```
However this service seems to be available in ROS and the device does react on input from the rqt reconfigure desktop app `rosrun rqt_reconfigure rqt_reconfigure`.