Intermodalics/tango_ros #186: Make basic tutorial rviz tutorial

PerrineAguiar, 2017-03-16:


PerrineAguiar, 2017-03-20:
I added the tutorial here: http://wiki.ros.org/tango_ros_streamer/Visualisation
@smits @jubeira @adamantivm @juergensturm Any comments to improve it is welcomed.

juergensturm, 2017-04-06:
Super cool, I love this. It's also great to see that the point cloud overlay to images works :+1: 

Just one very minor suggestion:
Although all the information is provided, people might find it hard to re-create the settings in RViz. Do you think you could summarize the most important ones (and for support, offer them the rviz file)? And/or I would suggest to unfold the Rviz settings so that people can see them in the screenshot (the config on the left, showing the properties of the relevant displays).

[image]

PerrineAguiar, 2017-04-07:
@juergensturm Thanks for your suggestion. We already give a link to the rviz file, but it is good idea to unfold the Rviz settings. I'll update the picture.

closed 2017-04-11 by PerrineAguiar