Intermodalics/tango_ros #188: Failed to built App from source

ychenco, 2017-03-18:
I followed the tutorial to build App from source and got the following error. 

**In "tango_ros_node.h", it includes "tango_ros_native/PublisherConfig.h" , I have checked if the file "PublisherConfig.h" exists, but found nothing.** 

Do you have any idea?
> ~/tango_ros_ws/src/tango_ros/TangoRosStreamer/tango_ros_node/src/main/jni/../../../../../tango_ros_common/tango_ros_native/../../third_party/miniglog/Android.mk:5: "Build miniglog.so/miniglog.a"
> ~/tango_ros_ws/src/tango_ros/TangoRosStreamer/tango_ros_node/src/main/jni/../../../../../tango_ros_common/tango_ros_native/../../third_party/miniglog/Android.mk:8: ~/tango_ros_ws/src/tango_ros/TangoRosStreamer/tango_ros_node/src/main/jni/../../../../../tango_ros_common/tango_ros_native/../../third_party/miniglog
> 
> [armeabi-v7a] Compile++ thumb: tango_ros_node <= jni.cc
> 
> In file included from ~/tango_ros_ws/src/tango_ros/TangoRosStreamer/tango_ros_node/src/main/jni/../../../../../tango_ros_common/tango_ros_native/include/tango_ros_native/tango_ros_util.h:20:0,
>                  from ~/tango_ros_ws/src/tango_ros/TangoRosStreamer/tango_ros_node/src/main/jni/jni.cc:16:
> ~/tango_ros_ws/src/tango_ros/TangoRosStreamer/tango_ros_node/src/main/jni/../../../../../tango_ros_common/tango_ros_native/include/tango_ros_native/tango_ros_node.h:41:46: fatal error: tango_ros_native/PublisherConfig.h: No such file or directory
>  #include "tango_ros_native/PublisherConfig.h"
>                                               ^
> compilation terminated.
> make: *** 
>
> [~/tango_ros_ws/src/tango_ros/TangoRosStreamer/tango_ros_node/src/main/obj/local/armeabi-v7a/objs/tango_ros_node/jni.o]
> Error1

smits, 2017-03-20:
That header is typically generated during the catkin build step of the tango_ros_native package.  Can you run `catkin build tango_ros_native` and check if that header is generated at `tango_ros_common/tango_ros_native/include/tango_ros_native/PublisherConfig.h`

ychenco, 2017-03-20:
Yeah, you are right. After installing opencv3 and building glog_catkin, I generated the head file and successfully built the app. It works well. Thanks !

closed 2017-03-21 by smits