Intermodalics/tango_ros #189: failed build with android studio: <ros/ros.h> not found

fishcu, 2017-03-19:
Hi!

I'm following the build from source tutorial in the readme.
Everything works fine up to the point that I try to build the app. The build from Android Studio fails with the following error in the log:
`[armeabi-v7a] Compile++ thumb: tango_ros_node <= jni.cc

In file included from /home/visionlab/dev/tango_ros_ws/src/tango_ros/TangoRosStreamer/tango_ros_node/src/main/jni/jni.cc:16:0:
/home/visionlab/dev/tango_ros_ws/src/tango_ros/TangoRosStreamer/tango_ros_node/src/main/jni/../../../../../tango_ros_common/tango_ros_native/include/tango_ros_native/tango_ros_util.h:18:21: fatal error: ros/ros.h: No such file or directory
 #include <ros/ros.h>
                     ^
compilation terminated.`

Any ideas how to fix this? I tried messing around with the corresponding CMakeLists.txt, but everything seems fine (when compared to other ROS packages).

Thanks!

smits, 2017-03-20:
Are you building for indigo or kinetic? Which Ubuntu distro? Did you successfully ran the catkin build steps? I guess the ROS include path is not picked up correctly.

@PerrineAguiar Although it will probably not fix this issue it might be a good idea to add roscpp as a dependency in the package.xml of the TangoRosStreamer package.

PerrineAguiar, 2017-03-20:
@smits Just opened an issue to fix this.

fishcu, 2017-03-20:
@smits: You are right, I should have mentioned this. I'm on Ubuntu 16.04, on a relatively fresh install of ROS kinetic.

My bad, I had accidentally copied over the "share" folder instead of the "include" folder from the roscpp_ndk_android folder. This fixes the issue of <ros/ros.h> not being found.

I now ran into a different problem. Not sure if I should open a new issue, or if I should just ask here.

This is the same error whether I try to build with catkin or with android studio. The error is as follows:
`error: no matching function for call to ‘ros::NodeHandle::param(std::__cxx11::string&, int&, tango_ros_native::LocalizationMode)’`

The full error message reads (from catkin, but the one from Android studio is equivalent):
```
visionlab@ub16-04-aw17:~/dev/tango_ros_ws/src$ catkin build tango_ros_native
---------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/visionlab/dev/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/visionlab/dev/tango_ros_ws
---------------------------------------------------------------------------------
Source Space:       [exists] /home/visionlab/dev/tango_ros_ws/src
Log Space:          [exists] /home/visionlab/dev/tango_ros_ws/logs
Build Space:        [exists] /home/visionlab/dev/tango_ros_ws/build
Devel Space:        [exists] /home/visionlab/dev/tango_ros_ws/devel
Install Space:      [unused] /home/visionlab/dev/tango_ros_ws/install
DESTDIR:            [unused] None
---------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------------------------
[build] Found '6' packages in 0.0 seconds.                                                                    
[build] Package table is up to date.                                                                          
Starting  >>> catkin_simple                                                                                   
Starting  >>> tango_client_api                                                                                
Finished  <<< catkin_simple                   [ 0.1 seconds ]                                                 
Starting  >>> glog_catkin                                                                                     
Finished  <<< tango_client_api                [ 0.2 seconds ]                                                 
Finished  <<< glog_catkin                     [ 0.2 seconds ]                                                 
Starting  >>> tango_ros_native                                                                                
______________________________________________________________________________________________________________
Errors     << tango_ros_native:make /home/visionlab/dev/tango_ros_ws/logs/tango_ros_native/build.make.002.log 
/home/visionlab/dev/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp: In member function ‘TangoErrorType tango_ros_native::TangoRosNode::TangoSetupConfig()’:
/home/visionlab/dev/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp:434:110: error: no matching function for call to ‘ros::NodeHandle::param(std::__cxx11::string&, int&, tango_ros_native::LocalizationMode)’
   node_handle_.param(publisher_config_.localization_mode_param, localization_mode, LocalizationMode::ODOMETRY);
                                                                                                              ^
In file included from /opt/ros/kinetic/include/ros/ros.h:45:0,
                 from /opt/ros/kinetic/include/camera_info_manager/camera_info_manager.h:41,
                 from /home/visionlab/dev/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_native/include/tango_ros_native/tango_ros_node.h:28,
                 from /home/visionlab/dev/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp:14:
/opt/ros/kinetic/include/ros/node_handle.h:2040:8: note: candidate: template<class T> bool ros::NodeHandle::param(const string&, T&, const T&) const
   bool param(const std::string& param_name, T& param_val, const T& default_val) const
        ^
/opt/ros/kinetic/include/ros/node_handle.h:2040:8: note:   template argument deduction/substitution failed:
/home/visionlab/dev/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp:434:110: note:   deduced conflicting types for parameter ‘const T’ (‘int’ and ‘tango_ros_native::LocalizationMode’)
   node_handle_.param(publisher_config_.localization_mode_param, localization_mode, LocalizationMode::ODOMETRY);
                                                                                                              ^
In file included from /opt/ros/kinetic/include/ros/ros.h:45:0,
[... cut ...]
[build] Summary: 3 of 4 packages succeeded.                                                                   
[build]   Ignored:   2 packages were skipped or are blacklisted.                                              
[build]   Warnings:  None.                                                                                    
[build]   Abandoned: None.                                                                                    
[build]   Failed:    1 packages failed.                                                                       
[build] Runtime: 3.8 seconds total.   
```

smits, 2017-03-21:
Hmm, that's weird, I cannot reproduce your error, are you on latest master? Did you build from a clean workspace?

PerrineAguiar, 2017-03-21:
I think this error was introduced some days ago in master, it should be fixed now. Pulling the latest master should fix the issue.

jubeira, 2017-03-21:
@fishcu yes, as @PerrineAguiar indicates it was fixed in #172 . Try pulling the latest master :+1: 
If you try building it with catkin, remember to use `catkin build --no-jobserver`

fishcu, 2017-03-21:
Thank you everyone, I can now build successfully! :+1:

closed 2017-03-22 by smits