Intermodalics/tango_ros #196: "Unfortunately, Tango Ros Streamer has stopped."

fishcu, 2017-03-23:
I'm building the app from source using Android studio, latest commit. Ubuntu 16. ROS kinetic.
I'm getting the problem that the app crashes often on the Android device -- I have the Google Tango tablet.

I cannot reproduce the crash exactly, but it only happens when the device is already hooked up to the network and is basically ready to start publishing data.
The behavior is usually that I start the app for the first time, the app crashes. I start the app a second time, it works as expected.
Here is a logcat that I tried to crop to the relevant parts (before app launch/after app crash). I don't know anything about Android programming, but it seems like a start for debugging:
http://pastebin.com/iCjU0ETw
Have you experienced similar things? Any ideas what might be causing this?

Cheers
Isaac

PerrineAguiar, 2017-03-24:
@fishcu Thanks for reporting this problem.
We notice this behavior when:
* the user kill the app by swapping it from the current app BUT does not press the shutdown button in the toolbar
or the other way around
* the user kill the app by pressing the shutdown button in the toolbar BUT does not swap it from the current app
So it seems that both actions are needed: first pressing shutdown, then swapping the app.
We are currently working on fixing this issue.

Could it be the reason of your crash?

fishcu, 2017-04-28:
Thank you for the hint that I also need to kill the app from the recent apps. As I noticed, I was receiving data from tango ros streamer even after I pressed "Shutdown". 

However, the app will still usually crash on the first launch, even on a freshly rebooted device. More than happy to provide more debug info if you tell me what you can use!