Intermodalics/tango_ros #283: Configurable Alignment frame for pointcloud (Question; not bug)

eric-schleicher, 2017-05-19:
What is the alignment frame of the pointcloud2 coming out of tango_ros?  and is this configurable by file value, service call or other method?

As I understand: The tango API provides onPointCloudAvailable provide a basic cloud with no transform applied to it.  typically applications then request a (1) transform based on the device extrinsics and then another (2) for time accurate pose (from start_of_service?)

Is it possible to get a cloud that is just the basic cloud; one that doesn't have any of the transforms?

In my application I'm using an external tracking system for odometry and i'm starting to think that i have to back out all (negate) of these transforms, but it seems problematic to do so.

Or is there a different approach (like consuming raw depth maps) that would make more sense.

Still new to ROS and tango_ros, so forgive (& inform) my ignorance on whether this is something that can/should be achieved by configuration in tango_ros, or something outside in ROS.

PerrineAguiar, 2017-06-19:
The point cloud is expressed in the "_camera_depth_" frame. This frame is part of our [tf](http://wiki.ros.org/tf) tree, which is basically: 
start_of_service --> device --> camera_depth.

What do you mean by configurable alignment frame? Do you mean that you would like to use a custom name for this frame? If so you should take a look at ROS [remapping](http://wiki.ros.org/roslaunch/XML/remap) and [tf_remap](http://answers.ros.org/question/216582/how-to-use-tf_remap/).
We are also considering adding ROS parameters to the Tango ROS node to configure frames names.

eric-schleicher, 2017-06-29:
I've learned a bit since writing this.  The easiest way to describe it is that I'm not using the start_of_service or device frames.   I have an external tracking system with much more accurate/faster pose than the start_of_service->device TF.   I had to make a static transform to map the device/camera_depth offset to my own tracking device TF.   

I think we can close this.

closed 2017-06-29 by eric-schleicher