Intermodalics/tango_ros #288: Allow to configure topic names in tango_ros_node

adamantivm, 2017-05-29:
When integrating tango_ros_node in another rosjava application (i.e. Tangobot) it is currently not possible to configure topic names, remappings or namespaces.

it would be very valuable to extend tango_ros_node to take in topic remapping arguments in general from the rosjava side and pass them through to the native node

PerrineAguiar, 2017-06-08:
@adamantivm Can you provide a pointer on existing way to do this in ROSJava?

jubeira, 2017-06-08:
The `NativeNodeMain`'s `execute` has a parameter for remapping arguments: https://github.com/jubeira/rosjava_core/blob/8c895e48799509e6792f0323b473de5dcb31e155/rosjava/src/main/java/org/ros/node/NativeNodeMain.java#L82
Those arguments have to be fetched from the native side off the application via JNI.
I don't know exactly where they have to be used and how, but I see that when you load the nodelet, an empty map<string, string> is sent as argument. You probably need to fill that one.
Hope it helps for a start!

adamantivm, 2017-06-08:
+1 thanks @jubeira that's exactly what I would have said

PerrineAguiar, 2017-06-08:
@jubeira Thanks for the pointer and the explanation!

PR #324 (2017-07-18): Add support for remmaping arguments
Fixes #288.
closed 2017-07-18 by PerrineAguiar