Intermodalics/tango_ros #299: Timestamp issue between tango and ros

PerrineAguiar, 2017-05-30:


adamantivm, 2017-05-30:
I can attest that when using tango_ros_node to publish TF in combination with other nodes publishing TF in different systems (from another computer, or even from Java in the same device), the TF transform timestamps published by the tango_ros_node differ with the rest of the system.
This prevents TF clients from properly resolving the TF tree and makes it impossible to properly configure a navigation stack.

PerrineAguiar, 2017-05-31:
@adamantivm @juergensturm @smits 

The issue comes from [this line](https://github.com/Intermodalics/tango_ros/blob/master/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp#L268). Here we calculate the offset between the ros time and the tango time. 
This calculation is actually not correct because the pose timestamp corresponds to the time of the last available pose returned by `TangoService_getPoseAtTime`, which has a small delayed compared to `ros::Time::now()`.

From [this post]( https://stackoverflow.com/questions/28073242/how-to-get-current-timestamp-tango-sdk), using `TangoService_getPoseAtTime` seems to be the only way to get time from Tango, but it does not allow to retrieve the current time, only the "most recent time".

After testing the app with the navigation stack and the turtlebot, it seems that not calculating the offset and setting message stamps to `ros::Time::now()` is solving the issue. This means that message stamps will correspond to the time when we enter Tango callbacks. This is not fully correct, if you look at [this post](https://stackoverflow.com/questions/28070864/exactly-how-do-we-compute-timestamp-differentials), the time when we enter the callback is not the same as the actual data timestamp.

PR #312 (2017-06-19): Fix/ros vs tango timestamp
This fixes #299.
closed 2017-06-19 by PerrineAguiar