Intermodalics/tango_ros #318: Replace onFrameAvailabe by onImageAvailable

smits, 2017-06-19:


PerrineAguiar, 2017-08-01:
Issue encountered:
* Calling `TangoService_connectOnImageAvailable` returns with the error ` Cannot register onImageAvailable callback when onFrameAvailable already registered!`, even after commenting the code which register onFrameAvailable.
Issue reported [here](https://stackoverflow.com/questions/44937974/invalid-input-result-when-calling-tangoservice-connectonimageavailable), but no answered yet.
UPDATE: Ignoring the error code seems to work.

* Cannot register the color camera with both onFrameAvailable and onImageAvailable (but no error code in this case...), while we still need to use onFrameAvailable because the new struct `TangoImage` is not compatible with the Tango 3d reconstruction API and tango support.

PerrineAguiar, 2017-08-10:
Test results:

* TangoService_connectOnFrameAvailable with camera id TANGO_CAMERA_FISHEYE throws an error on the Asus phone. The callback is not triggered.
The error message is:
Cannot register onFrameAvailable callback when onImageAvailable already registered!

* TangoService_connectOnImageAvailable with camera id TANGO_CAMERA_COLOR (only the color camera is supported for onImageAvailable according to the documentation), throws an error on both the Asus and Lenovo phone. The callback is still triggered.
The error message is:
Cannot register onImageAvailable callback when onFrameAvailable already registered!

PerrineAguiar, 2017-08-10:
This is not solving #314. The reason of #314 and its fix is in #371.