Intermodalics/tango_ros #336: Using tango_ros_node

hungbv11, 2017-06-29:
Hey guys,

I am planning to using tango_ros_node for an app in my master thesis. Basically, in the app, the tango device will stream image data to a computer to do some machine learning and then stream the processed data back and show it on the tango. Is it possible to do that? 
I am quite a newbie in ros_android and may need your helps.
Thanks a lot,

PerrineAguiar, 2017-06-29:
Hi @hungbv11 ,

Great to see that you are interested in the tango_ros_node!
With this node you can stream Tango data to a computer using ROS, it should be quite straight forward. Here are some useful links:
[ROS wiki](http://wiki.ros.org/tango_ros_streamer)
[Tutorial to use the tango ROS node in your own app](http://wiki.ros.org/tango_ros_streamer/tango_ros_node)
Then it should be possible to process data as you like and stream it back to the device for display.

Feel free to post questions/issues if you have any.
Cheers,
Perrine

adamantivm, 2017-06-29:
You can also look at https://github.com/ekumenlabs/tangobot as another example project that uses tango_ros_node.

smits, 2017-07-07:
I consider this issue closed ;)

closed 2017-07-07 by smits