Intermodalics/tango_ros #343: Frame rate very slow?

trinamntn08, 2017-07-11:
I want to record Image_Raw from Project Tango into a bag by using the command below:
rosbag record /tango/camera/color_1/image_raw  
i got a bag with sequence of image_raw. BUt once, i visualize on rviz, i realize the frame rate is really slow.
Is there any method to verify frame rate to record into the bag , and how can we change this rate ?

PerrineAguiar, 2017-07-12:
You can check the rate of a topic with the following command: `topic hz <topic_name>`.
Unfortunately there is no way to change the rate of the color image topic, it should be around 1Hz which is very slow. The main reason is the big size of the image (1920*1080) which leads to:
* slow serialization/deserialization of the image message
* slow transport of the image message over wifi

To increase the rate, you could try:
* using USB tethering with your Tango device instead of wifi
* recording the compressed image instead (`/tango/camera/color_1/image_raw/compressed`). Its rate should be between 5 and 10 Hz. Note that you can change the compressing format/level using [dynamic reconfigure](http://wiki.ros.org/dynamic_reconfigure), as explained at the end of [this tutorial](http://wiki.ros.org/tango_ros_streamer/Visualisation).

trinamntn08, 2017-07-12:
Is there any chance that i can  exceed more than 20 Hz ?

PerrineAguiar, 2017-07-13:
I don't think you will be able to exceed 20Hz for the color camera topic, even using the compressed version.
If slow rate is really an issue for you, you will need to figure out how to record a rosbag directly on device. That way, using the device as master, you would avoid latency due to wifi transport. As mentioned in #170, having a rosbag recorder on device is not on our roadmap.

eric-schleicher, 2017-07-14:
@PerrineAguiar usb connection is possible (physically) in my and would be advantageous. Is there information somewhere on how to run in a tethered configuration?  this is the first time i've seen it referenced.

PerrineAguiar, 2017-07-14:
@eric-schleicher To enable USB tethering on your device, plug the device to your desktop via USB and go to: Settings --> Wireless & networks --> Tethering & portable hotspot --> enable USB tethering.
On your desktop, you should then be able to connect to this new wired connection. Still on your desktop, run 'roscore'.
On your device start the TangoROSStreamer app, go to the settings and set the new master uri as: `http://<ip_address>:11311`
To find the new ip address of your desktop, you can run `ifconfig` and search for an interface called `usb0`. The latter corresponds to your tethered network and there you should see the correct ip address.

Note that for any new terminal where you want to communicate with the tango node, you will have to set the ROS_MASTER_URI with this ip address using the following command:
`export ROS_MASTER_URI=http://<ip_address>:11311`

For more information about ROS network configuration, yo can check [this tutorial](http://wiki.ros.org/ROS/NetworkSetup).

eric-schleicher, 2017-07-14:
Thank you very much!  

Awesome :)

trinamntn08, 2017-07-17:
@eric-schleicher  : can you tell me how to record raw images via USB connection, pls ?

eric-schleicher, 2017-07-25:
Sorry @trinamntn08  I'm not sure how to do that.  you might be able to find a ROS node that subscribes to a topic with RGB images and writes them to disk.

closed 2017-10-02 by PerrineAguiar