Intermodalics/tango_ros #348: rosbag record /tango/camera/color_1/image_raw  issue???

trinamntn08, 2017-07-12:
I have an issue when using Tango Ros Streamer to record Image_raw to create my own data set. 
When i run the commend below:
rosbag record /tango/camera/color_1/image_raw 
i recognized that the frame recorded rate is really slow, and image_raw is not colored , just grey image. because of that,  ORB-SLAM program can not be initialized 
Do you have any ideal to resolve this issue ? or i have done something wrong ?
 I really appreciate your help ^^

trinamntn08, 2017-07-12:
In fact, when i run rviz to visualize colored image from /tango/camera/color_1/image_raw, i see that camera works perfectly with colored image. Just when i start recording image_raw, it appears the problem.

PerrineAguiar, 2017-07-13:
How do you visualize the recorded color image message?
If you can visualize it in color with rviz while streaming, you should be able to do the same while playing your rosbag.

closed 2017-10-02 by PerrineAguiar