Intermodalics/tango_ros #357: App build fails

furqan-habibi, 2017-07-20:
I have done all the build steps as written, but end up with a build fails. I have build successfully in the past (~version 1.1).
The error messages are as follows:
```
*
*
*
[1m> Building 25% > :tango_ros_common:mergeReleaseResources[22m[0K[56D:tango_ros_common:mergeReleaseResources[0K
[1m> Building 25%[22m[0K[14D:tango_ros_common:processReleaseManifest
[1m> Building 26% > :tango_ros_common:processReleaseResources[22m[58D:tango_ros_common:processReleaseResources[0K
[1m> Building 26%[22m[0K[14D:tango_ros_common:generateReleaseSources
:tango_ros_common:incrementalReleaseJavaCompilationSafeguard
:tango_ros_common:javaPreCompileRelease
:tango_ros_common:compileReleaseJavaWithJavac
:tango_ros_common:compileReleaseJavaWithJavac - is not incremental (e.g. outputs have changed, no previous execution, etc.).
[1m> Building 27% > :tango_ros_common:compileReleaseJavaWithJavac[22m[62D/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:36: error: cannot find symbol
[1m> Building 27% > :tango_ros_common:compileReleaseJavaWithJavac[22m[62Dimport tango_ros_messages.LoadOccupancyGrid;[0K
                         ^
  symbol:   class LoadOccupancyGrid
  location: package tango_ros_messages
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:37: error: cannot find symbol
import tango_ros_messages.LoadOccupancyGridRequest;
                         ^
  symbol:   class LoadOccupancyGridRequest
  location: package tango_ros_messages
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:38: error: cannot find symbol
import tango_ros_messages.LoadOccupancyGridResponse;
                         ^
  symbol:   class LoadOccupancyGridResponse
  location: package tango_ros_messages
[1m> Building 27% > :tango_ros_common:compileReleaseJavaWithJavac[22m[62D/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:198: error: cannot find symbol
        saveMapRequest.setRequest((byte)(saveMapRequest.SAVE_LOCALIZATION_MAP | saveMapRequest.SAVE_OCCUPANCY_GRID));
                                                       ^
  symbol:   variable SAVE_LOCALIZATION_MAP
  location: variable saveMapRequest of type SaveMapRequest
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:198: error: cannot find symbol
        saveMapRequest.setRequest((byte)(saveMapRequest.SAVE_LOCALIZATION_MAP | saveMapRequest.SAVE_OCCUPANCY_GRID));
                                                                                              ^
  symbol:   variable SAVE_OCCUPANCY_GRID
  location: variable saveMapRequest of type SaveMapRequest
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:198: error: cannot find symbol
        saveMapRequest.setRequest((byte)(saveMapRequest.SAVE_LOCALIZATION_MAP | saveMapRequest.SAVE_OCCUPANCY_GRID));
                      ^
  symbol:   method setRequest(byte)
  location: variable saveMapRequest of type SaveMapRequest
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:204: error: cannot find symbol
                        saveMapResponse.getMessage(), saveMapResponse.getLocalizationMapName(),
                                                                     ^
  symbol:   method getLocalizationMapName()
  location: variable saveMapResponse of type SaveMapResponse
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:205: error: cannot find symbol
                        saveMapResponse.getLocalizationMapUuid());
                                       ^
  symbol:   method getLocalizationMapUuid()
  location: variable saveMapResponse of type SaveMapResponse
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:217: error: cannot find symbol
        ServiceClient<LoadOccupancyGridRequest, LoadOccupancyGridResponse> loadOccupancyGridService =
                      ^
  symbol:   class LoadOccupancyGridRequest
  location: class TangoServiceClientNode
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:217: error: cannot find symbol
        ServiceClient<LoadOccupancyGridRequest, LoadOccupancyGridResponse> loadOccupancyGridService =
                                                ^
  symbol:   class LoadOccupancyGridResponse
  location: class TangoServiceClientNode
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:220: error: cannot find symbol
                        LoadOccupancyGrid._TYPE);
                        ^
  symbol:   variable LoadOccupancyGrid
  location: class TangoServiceClientNode
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:222: error: cannot find symbol
        LoadOccupancyGridRequest loadOccupancyGridRequest = mConnectedNode.getServiceRequestMessageFactory().newFromType(LoadOccupancyGrid._TYPE);
        ^
  symbol:   class LoadOccupancyGridRequest
  location: class TangoServiceClientNode
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:222: error: cannot find symbol
        LoadOccupancyGridRequest loadOccupancyGridRequest = mConnectedNode.getServiceRequestMessageFactory().newFromType(LoadOccupancyGrid._TYPE);
                                                                                                                         ^
  symbol:   variable LoadOccupancyGrid
  location: class TangoServiceClientNode
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:224: error: cannot find symbol
        loadOccupancyGridService.call(loadOccupancyGridRequest, new ServiceResponseListener<LoadOccupancyGridResponse>() {
                                                                                            ^
  symbol:   class LoadOccupancyGridResponse
  location: class TangoServiceClientNode
/home/kaigi/tango_ros_ws/src/tango_ros/tango_ros_common/tango_ros_common/src/main/java/eu/intermodalics/tango_ros_common/TangoServiceClientNode.java:226: error: cannot find symbol
            public void onSuccess(LoadOccupancyGridResponse loadOccupancyGridResponse) {
                                  ^
  symbol: class LoadOccupancyGridResponse
[1m> Building 27% > :tango_ros_common:compileReleaseJavaWithJavac[22m[62D15 errors[0K
:tango_ros_common:compileReleaseJavaWithJavac [33mFAILED[39m

[31mFAILURE: [39m[31mBuild failed with an exception.[39m

* What went wrong:
Execution failed for task ':tango_ros_common:compileReleaseJavaWithJavac'.
[33m> [39mCompilation failed; see the compiler error output for details.

* Try:
Run with [1m--stacktrace[22m option to get the stack trace. Run with [1m--info[22m or [1m--debug[22m option to get more log output.

[31mBUILD FAILED[39m

Total time: 46.431 secs
[0K[0mCMakeFiles/gradle-tango_ros_streamer.dir/build.make:57: recipe for target 'CMakeFiles/gradle-tango_ros_streamer' failed
make[2]: *** [CMakeFiles/gradle-tango_ros_streamer] Error 1
CMakeFiles/Makefile2:291: recipe for target 'CMakeFiles/gradle-tango_ros_streamer.dir/all' failed
make[1]: *** [CMakeFiles/gradle-tango_ros_streamer.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

It looks like some newer class (LoadOccupancyGrid etc.) in tango_ros_messages is not detected by the compiler. I am not quite sure but there might be a connection with the fact that tango_ros_messages maven repository still serving old (1.0.0) version.
https://github.com/rosjava/rosjava_mvn_repo/tree/master/org/ros/rosjava_messages/tango_ros_messages

furqan-habibi, 2017-07-21:
I finally managed to make it build successfully. The steps are as follows:
- I first tried to generate the java class from tango_ros_messages service files using genjava_message_artifacts. 
```
$ genjava_message_artifacts -p tango_ros_messages
```
- This in turn spurn out another error that says it need std_msgs-0.5.11, which is still not published in the rosjava maven repo. 
```
Generating message artifacts for: 
['tango_ros_messages']

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all dependencies for configuration ':runtime'.
> Could not find org.ros.rosjava_messages:std_msgs:0.5.11.
  Searched in the following locations:
      file:/home/kaigi/tango_ros_ws/devel/share/maven/org/ros/rosjava_messages/std_msgs/0.5.11/std_msgs-0.5.11.pom
      file:/home/kaigi/tango_ros_ws/devel/share/maven/org/ros/rosjava_messages/std_msgs/0.5.11/std_msgs-0.5.11.jar
```
- So I decided to download the 0.5.10 version and change the version in the manifest and the filename. I then put it in `devel/share/maven/org/ros/rosjava_messages/std_msgs/0.5.11` and build tango_ros_messages again.
- Then I try to build the app but it fails again, with error saying the symlink process did not go successfully.
```
Errors     << tango_ros_messages:symlink /home/kaigi/tango_ros_ws/logs/tango_ros_messages/build.symlink.000.log
Warning: Cannot symlink from /home/kaigi/tango_ros_ws/devel/.private/tango_ros_messages/share/maven/org/ros/rosjava_messages/tango_ros_messages/maven-metadata.xml to existing file /home/kaigi/tango_ros_ws/devel/share/maven/org/ros/rosjava_messages/tango_ros_messages/maven-metadata.xml
Warning: Source hash: 304162665fe48b6f38537f737a45475a
Warning: Dest hash: d3bbe18cef3ab68bd05b0bbdc9bf9b51
Warning: Cannot symlink from /home/kaigi/tango_ros_ws/devel/.private/tango_ros_messages/share/maven/org/ros/rosjava_messages/tango_ros_messages/maven-metadata.xml.sha1 to existing file /home/kaigi/tango_ros_ws/devel/share/maven/org/ros/rosjava_messages/tango_ros_messages/maven-metadata.xml.sha1
Warning: Source hash: 31fddb7cf15f9c659221d69f108b19dd
Warning: Dest hash: 290f3ece23c29c11ea9cbd5893bc0c67
```
- I then delete the generated tango_ros_messages jar file.
- After that I tried to build it again, and now it went successfully.

Though I can build the app successfully, this seems like a very hacky way to do it. If someone from the development team can help investigate this I'd be really thankful. However I know you guys are busy in the development cycle right now so keep up the hard work and I love what you guys are doing.

Cheers.

PerrineAguiar, 2017-07-24:
@furqan-habibi Thanks for the kind words.

I think your issue was indeed related to the new classes of the `tango_ros_messages` package. I guess what really solved the problem was deleting the `tango_ros_messages.jar` and rebuilding.

Today we released the new version of the `tango_ros_messages` package. So the app now depends on `org.ros.rosjava_messages:tango_ros_messages:2.0.0` (instead of 1.0.0). 
As a consequence, if you pull the latest changes from master, you might get some issues again with this dependency while building.
The solution we found for this problem is as follow:
* Remove devel and build folders
```
cd ~/tango_ros_ws/
rm -r build/ devel/
```
* Rebuild with catkin
```
catkin build --no-jobserver --verbose
```
* Exit Android Studio
* Source the devel folder
```
source ~/tango_ros_ws/devel/setup.bash
```
and in the same terminal restart Android Studio

I hope this helps.

closed 2017-10-02 by PerrineAguiar