Intermodalics/tango_ros #376: App crash when "unreliable" is checked in rviz for an image stream

PerrineAguiar, 2017-09-01:
When "unreliable" is checked UDP transport is used.

PerrineAguiar, 2017-09-01:
@meyerj FYI

meyerj, 2017-09-01:
Thanks for opening the issue!

Here is the related `adb logcat` output, even if the backtrace does not seem to be very useful:
```
--------- beginning of crash
08-23 21:58:27.576 11674 11780 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x1c in tid 11780 (Thread-161)
08-23 21:58:27.641   510   510 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
08-23 21:58:27.641   510   510 F DEBUG   : Build fingerprint: 'Lenovo/phinny_na/PB2PRO:6.0.1/MMB29M/PB2-690Y_S200021_161202:userdebug/release-keys'
08-23 21:58:27.641   510   510 F DEBUG   : Revision: '0'
08-23 21:58:27.641   510   510 F DEBUG   : ABI: 'arm'
08-23 21:58:27.642   510   510 F DEBUG   : pid: 11674, tid: 11780, name: Thread-161  >>> eu.intermodalics.tango_ros_streamer <<<
08-23 21:58:27.642   510   510 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x1c
08-23 21:58:27.652   510   510 I debuggerd: type=1400 audit(0.0:240): avc: denied { read } for name="libtango_client_api.so" dev="mmcblk0p48" ino=516904 scontext=u:r:debuggerd:s0 tcontext=u:object_r:app_data_file:s0:c512,c768 tclass=file permissive=1
08-23 21:58:27.652   510   510 I debuggerd: type=1400 audit(0.0:241): avc: denied { open } for path="/data/data/com.google.tango/libfiles/armeabi-v7a/libtango_client_api.so" dev="mmcblk0p48" ino=516904 scontext=u:r:debuggerd:s0 tcontext=u:object_r:app_data_file:s0:c512,c768 tclass=file permissive=1
08-23 21:58:27.652   510   510 I debuggerd: type=1400 audit(0.0:242): avc: denied { getattr } for path="/data/data/com.google.tango/libfiles/armeabi-v7a/libtango_client_api.so" dev="mmcblk0p48" ino=516904 scontext=u:r:debuggerd:s0 tcontext=u:object_r:app_data_file:s0:c512,c768 tclass=file permissive=1
08-23 21:58:27.698   510   510 F DEBUG   :     r0 d80eb198  r1 0000001c  r2 aae6dcb0  r3 00000000
08-23 21:58:27.699   510   510 F DEBUG   :     r4 0000001c  r5 d80eb198  r6 dbf98b84  r7 00000000
08-23 21:58:27.699   510   510 F DEBUG   :     r8 d80eb1e4  r9 d80eb2c0  sl 000005dc  fp dbf993d0
08-23 21:58:27.699   510   510 F DEBUG   :     ip dbf98bd8  sp d80eb178  lr dbb1a10d  pc dbaf7f26  cpsr 000f0030
08-23 21:58:27.705   489  4983 W qdmemalloc: Gralloc using 4K chunk.
08-23 21:58:27.710   510   510 F DEBUG   : 
08-23 21:58:27.710   510   510 F DEBUG   : backtrace:
08-23 21:58:27.710   510   510 F DEBUG   :     #00 pc 00237f26  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN5boost9function1IvRKNS_10shared_ptrIN3ros9TransportEEEE11move_assignERS7_+9)
08-23 21:58:27.710   510   510 F DEBUG   :     #01 pc 0025a109  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so
08-23 21:58:27.710   510   510 F DEBUG   :     #02 pc 0025ac8d  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN3ros10Connection10initializeERKN5boost10shared_ptrINS_9TransportEEEbRKNS1_8functionIFbRKNS2_IS0_EERKNS_6HeaderEEEE+144)
08-23 21:58:27.710   510   510 F DEBUG   :     #03 pc 00235103  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN3ros17ConnectionManager24udprosIncomingConnectionERKN5boost10shared_ptrINS_12TransportUDPEEERNS_6HeaderE+146)
08-23 21:58:27.710   510   510 F DEBUG   :     #04 pc 0022645f  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN3ros12TopicManager12requestTopicERKSsRN6XmlRpc11XmlRpcValueES5_+2442)
08-23 21:58:27.710   510   510 F DEBUG   :     #05 pc 00227289  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN3ros12TopicManager20requestTopicCallbackERN6XmlRpc11XmlRpcValueES3_+68)
08-23 21:58:27.710   510   510 F DEBUG   :     #06 pc 0022170d  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN5boost6detail8function26void_function_obj_invoker2INS_3_bi6bind_tIvNS_4_mfi3mf2IvN3ros12TopicManagerERN6XmlRpc11XmlRpcValueESB_EENS3_5list3INS3_5valueIPS8_EENS_3argILi1EEENSH_ILi2EEEEEEEvSB_SB_E6invokeERNS1_15function_bufferESB_SB_+16)
08-23 21:58:27.710   510   510 F DEBUG   :     #07 pc 00268965  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZNK5boost9function2IvRN6XmlRpc11XmlRpcValueES3_EclES3_S3_+16)
08-23 21:58:27.711   510   510 F DEBUG   :     #08 pc 0029bf79  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN6XmlRpc22XmlRpcServerConnection13executeMethodERKSsRNS_11XmlRpcValueES4_+24)
08-23 21:58:27.711   510   510 F DEBUG   :     #09 pc 0029cc25  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN6XmlRpc22XmlRpcServerConnection14executeRequestEv+56)
08-23 21:58:27.711   510   510 F DEBUG   :     #10 pc 0029bd27  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN6XmlRpc22XmlRpcServerConnection13writeResponseEv+18)
08-23 21:58:27.711   510   510 F DEBUG   :     #11 pc 0029be4f  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN6XmlRpc22XmlRpcServerConnection11handleEventEj+50)
08-23 21:58:27.711   510   510 F DEBUG   :     #12 pc 0029a3b5  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN6XmlRpc14XmlRpcDispatch4workEd+500)
08-23 21:58:27.711   510   510 F DEBUG   :     #13 pc 0026a499  /data/app/eu.intermodalics.tango_ros_streamer-1/lib/arm/libnodelet_manager.so (_ZN3ros13XMLRPCManager16serverThreadFuncEv+192)
[... cut ...]
08-23 21:58:29.352  5744  6273 I tango   : data_manager.cc:1090 Behind by: 1.0398 seconds, skip current frame.
08-23 21:58:29.359  5744  6273 I tango   : data_manager.cc:1090 Behind by: 0.848836 seconds, skip current frame.
08-23 21:58:29.366   536   536 I Zygote  : Process 11674 exited due to signal (11)
08-23 21:58:29.368  5744  6273 I tango   : data_manager.cc:1090 Behind by: 0.671294 seconds, skip current frame.
08-23 21:58:29.368  1491  4719 I ActivityManager: Process eu.intermodalics.tango_ros_streamer (pid 11674) has died
```