Intermodalics/tango_ros #383: Tango Ros Streamer For phab 2 pro

gizemarik, 2017-12-04:
hi,i want to show and edit my 3D simulation from Rtab-map on tango device to my desktop.While using my tango device everything is OK. i can create 3D simulation and can see datas and coordinates but i want to connect my device and my computer .i install tango ros streamer and in my devices, i fill the correct master url then ros light turn on green , but tango light is still yellow. i can start rviz and rtabmap in my desktop seperately however how can i connect tango device and my desktop 
my tango device is lenovo phab 2 pro , and im using ubuntu 16.04

PerrineAguiar, 2017-12-13:
Hi @gizemarik,
After setting the ROS Master URI you should be able to receive Tango data on your desktop (enable local master should not be checked). Your device should be in the same network as your desktop.
Here are some tutorials that could help: 
http://wiki.ros.org/tango_ros_streamer/Visualisation
http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer
http://wiki.ros.org/ROS/NetworkSetup

fangxu622, 2018-12-21:
@psoetens @smits  thanks all

fangxu622, 2019-01-06:
@PerrineAguiar  hi .now i'm have two questions~thanks your reply

first .i'm sure start a roscore in my desktop, and which address should be fill in the MASTER_URI of 
 the phone ,and fill 192.168.43.106/1/255 ,it's incorrect likely~,how solve this ,i want to run master in my desktop1

second. and run the roscore in the phone ,and in my desktop set export $ROS_MASTER_URI=192.168.43.1 $ROS_HOSTNAME=192.168.43.1 ,it run successfully,like this

and why the colorcamera gui display have other thing ,if i want save the color image raw.
[image]

PerrineAguiar, 2019-01-14:
Hi @fangxu622 
The master URI to fill in the app should be of the following format: `http://<ip_of_your_desktop>:11311`.
Concerning the color image display, you are using a `Camera` display which means 3D objects (e.g. pointcloud) are projected in the image. If you want the color image only, you can use the `Image` display.

fangxu622, 2019-01-14:
> Hi @fangxu622
> The master URI to fill in the app should be of the following format: `http://<ip_of_your_desktop>:11311`.
> Concerning the color image display, you are using a `Camera` display which [...]

yes ，I ‘m sorry for making a spelling mistake. i fill app in `http://192.168.43.106:11311`., tango phone and desktop connect same network ,i want to make the desktop as a master,and  i fill `http://192.168.43.106:11311` in app . it's incorrect ...

you can see this .

> @gizemarik @PerrineAguiar ，hi~I get some problem ,my desktop don't connect the tango device ~my device is lenovo phab 2 pro ~~，i install tango-ros-streamer app version 1.3.1
> 
> my desktop environment ubuntu 16.04 kinetic~~
> 
> my desktop connect phab-2-pro wlan hotspot ,it make them on the same network .like this
> [image]
> 
> but on my tango device like this
> 
> [image]
> 
> my device info
> [image]
> 
> I'm very appreciate for any guidance

swiss-knight, 2019-02-21:
I also cannot connect to the master URI of ros from the smartphone "Tango ROS Streamer" app.
`roscore` is launched on the laptop and I can successfully ping the laptop's local net IP from the phone terminal.