Intermodalics/tango_ros #384: ROS connects but nodes do not

dan9thsense, 2018-01-07:
I also posted this at
https://answers.ros.org/question/279027/tango-ros-streamer-fail/

I installed tango_ros_streamer from the play store, following the instructions here: http://wiki.ros.org/tango_ros_streamer

The app loaded and connected to the roscore on my main computer and the ROS indicator turned green. However, the tango indicator is yellow and I am not seeing any data other than in the /android/IMU topic, which is publishing normally.

Here are the topics that exist:

/android/imu

/rosout

/rosout_agg

/tango/status
Here are the nodes that exist:

/android

/parameter_node

/rosout

/tango_service_client_node
Here is the tango log file (192.168.1.62 is the correct address for the tango):

> ed: Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</parameter_node,
> http://192.168.1.62:33283/>,
> TopicIdentifier</rosout>>,
> Topic<TopicIdentifier</rosout>,
> TopicDescription<rosgraph_msgs/Log,
> acffd30cd6b6de30f120938c17c593fb>>>>
> I/Registrar( 4394): Response<Success,
> Success, []> I/Registrar( 4394):
> Response<Success, Success, []>
> I/DefaultPublisher( 4394): Publisher
> registered:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</tango_service_client_node,
> http://192.168.1.62:38113/>,
> TopicIdentifier</rosout>>,
> Topic<TopicIdentifier</rosout>,
> TopicDescription<rosgraph_msgs/Log,
> acffd30cd6b6de30f120938c17c593fb>>>>
> I/DefaultPublisher( 4394): Publisher
> registered:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</android,
> http://192.168.1.62:56484/>,
> TopicIdentifier</rosout>>,
> Topic<TopicIdentifier</rosout>,
> TopicDescription<rosgraph_msgs/Log,
> acffd30cd6b6de30f120938c17c593fb>>>>
> I/Registrar( 4394): Registering
> subscriber:
> Subscriber<Topic<TopicIdentifier</tango/status>,
> TopicDescription<std_msgs/Int8,
> 27ffa0c9c4b8fb8492252bcad9e5c57b>>>
> I/Registrar( 4394): Registering
> publisher:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</android,
> http://192.168.1.62:56484/>,
> TopicIdentifier</android/imu>>,
> Topic<TopicIdentifier</android/imu>,
> TopicDescription<sensor_msgs/Imu,
> 6a62c6daae103f4ff57a132d6f95cec2>>>>
> I/Registrar( 4394): Response<Success,
> Success, []> I/Registrar( 4394):
> Response<Success, Success, []>
> I/DefaultPublisher( 4394): Publisher
> registered:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</android,
> http://192.168.1.62:56484/>,
> TopicIdentifier</android/imu>>,
> Topic<TopicIdentifier</android/imu>,
> TopicDescription<sensor_msgs/Imu,
> 6a62c6daae103f4ff57a132d6f95cec2>>>>
> I/RunningActivity( 5105):
> initAndStartRosJavaNode
> I/RunningActivity( 5105): Waiting for
> ROS CONNECTION latch release...
> I/RunningActivity( 5105): ROS
> CONNECTION latch released!
> E/RunningActivity( 5105): Uncaught
> exception of type class
> java.lang.RuntimeException
> I/Registrar( 5105):
> MasterXmlRpcEndpoint URI:
> http://192.168.1.62:11311/
> I/Registrar( 5105):
> MasterXmlRpcEndpoint URI:
> http://192.168.1.62:11311/
> I/Registrar( 5105):
> MasterXmlRpcEndpoint URI:
> http://192.168.1.62:11311/
> I/Registrar( 5105): Registering
> publisher:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</android,
> http://192.168.1.62:57114/>,
> TopicIdentifier</rosout>>,
> Topic<TopicIdentifier</rosout>,
> TopicDescription<rosgraph_msgs/Log,
> acffd30cd6b6de30f120938c17c593fb>>>>
> I/Registrar( 5105): Registering
> publisher:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</tango_service_client_node,
> http://192.168.1.62:55304/>,
> TopicIdentifier</rosout>>,
> Topic<TopicIdentifier</rosout>,
> TopicDescription<rosgraph_msgs/Log,
> acffd30cd6b6de30f120938c17c593fb>>>>
> I/Registrar( 5105): Registering
> publisher:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</parameter_node,
> http://192.168.1.62:51005/>,
> TopicIdentifier</rosout>>,
> Topic<TopicIdentifier</rosout>,
> TopicDescription<rosgraph_msgs/Log,
> acffd30cd6b6de30f120938c17c593fb>>>>
> I/Registrar( 5105): Response<Success,
> Success, []> I/Registrar( 5105):
> Response<Success, Success, []>
> I/Registrar( 5105): Response<Success,
> Success, []> I/DefaultPublisher(
> 5105): Publisher registered:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</parameter_node,
> http://192.168.1.62:51005/>,
> TopicIdentifier</rosout>>,
> Topic<TopicIdentifier</rosout>,
> TopicDescription<rosgraph_msgs/Log,
> acffd30cd6b6de30f120938c17c593fb>>>>
> I/DefaultPublisher( 5105): Publisher
> registered:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</android,
> http://192.168.1.62:57114/>,
> TopicIdentifier</rosout>>,
> Topic<TopicIdentifier</rosout>,
> TopicDescription<rosgraph_msgs/Log,
> acffd30cd6b6de30f120938c17c593fb>>>>
> I/DefaultPublisher( 5105): Publisher
> registered:
> Publisher<PublisherDefinition<PublisherIdentifier<NodeIdentifier</tango_service_client_node,
> http://192.168.1.62:55304/>,
> TopicIdentifier</rosout>>,
> Topic<TopicIdentifier</rosout>,
> TopicDescription<rosgraph_msgs/Log,
> acffd30cd6b6de30f120938c17c593fb>>>>
> I/Registrar( 5105): Registering
> publisher:
> Publisher<PublisherDefinition<PublisherIdentifier ...

PerrineAguiar, 2018-01-16:
Hi @dan9thsense 
Which device are you using?
Are you using the latest version of the app (v1.3.1)?
What are the settings of the app?
It seems that Tango is not starting properly, is it staying orange forever or does it turn red at one point?

dan9thsense, 2018-01-23:
Sorry for the delay in responding-- I was traveling.
I am using the Tango Development device, Android version 4.4.2, Kernal version 3.10.24-gd25b167, build number KOT49H.161129
I am using the app version 1.3.1
Settings:
Run local master is OFF
Master URI is correct, as I am able to connect to the master on the laptop (http://192.168.1.228:11311)
Create a new localization map is ON
Enable depth camera is ON
Enable color camera is OFF
Autostart is OFF

The ROS light starts red and turns green.
The Tango light stays yellow forever.

Appreciate the help.

PerrineAguiar, 2018-01-26:
Hi
Thanks for giving more details.

I could reproduce your issue on a Tango Development device.
Here is the output of `adb logcat`:

```
I/master_checker( 8622): Connected!
I/RunningActivity( 8622): ROS CONNECTION latch released!
I/TangoInitializationHelp( 8622): basePath: /data/data/com.google.tango/libfiles/
E/art     ( 8622): dlopen("/data/data/com.google.tango/libfiles/arm64-v8a/libtango_client_api.so", RTLD_LAZY) failed: dlopen failed: library "/data/data/com.google.tango/libfiles/arm64-v8a/libtango_client_api.so" not found
E/TangoInitializationHelp( 8622): java.lang.UnsatisfiedLinkError: dlopen failed: library "/data/data/com.google.tango/libfiles/arm64-v8a/libtango_client_api.so" not found
I/TangoInitializationHelp( 8622): Success! Using armeabi-v7a/libtango_client_api.
E/art     ( 8622): dlopen("/data/app-lib/eu.intermodalics.tango_ros_streamer-1/libnodelet_manager.so", RTLD_LAZY) failed: dlopen failed: could not load library "libtango_3d_reconstruction.so" needed by "libnodelet_manager.so"; caused by cannot locate symbol "rand" referenced by "libtango_3d_reconstruction.so"...
E/RunningActivity( 8622): Uncaught exception of type class java.lang.RuntimeException
W/System.err( 8622): java.lang.RuntimeException: An error occured while executing doInBackground()
W/System.err( 8622): 	at android.os.AsyncTask$3.done(AsyncTask.java:300)
W/System.err( 8622): 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
W/System.err( 8622): 	at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
W/System.err( 8622): 	at java.util.concurrent.FutureTask.run(FutureTask.java:242)
W/System.err( 8622): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:231)
W/System.err( 8622): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
W/System.err( 8622): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
W/System.err( 8622): 	at java.lang.Thread.run(Thread.java:811)
W/System.err( 8622): Caused by: java.lang.UnsatisfiedLinkError: dlopen failed: could not load library "libtango_3d_reconstruction.so" needed by "libnodelet_manager.so"; caused by cannot locate symbol "rand" referenced by "libtango_3d_reconstruction.so"...
W/System.err( 8622): 	at java.lang.Runtime.loadLibrary(Runtime.java:364)
W/System.err( 8622): 	at java.lang.System.loadLibrary(System.java:526)
W/System.err( 8622): 	at eu.intermodalics.nodelet_manager.TangoInitializationHelper.loadTangoRosNodeSharedLibrary(TangoInitializationHelper.java:190)
W/System.err( 8622): 	at eu.intermodalics.tango_ros_streamer.activities.RunningActivity.init(RunningActivity.java:764)
W/System.err( 8622): 	at eu.intermodalics.tango_ros_streamer.activities.RunningActivity$17.doInBackground(RunningActivity.java:865)
W/System.err( 8622): 	at eu.intermodalics.tango_ros_streamer.activities.RunningActivity$17.doInBackground(RunningActivity.java:862)
W/System.err( 8622): 	at android.os.AsyncTask$2.call(AsyncTask.java:288)
W/System.err( 8622): 	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
W/System.err( 8622): 	... 4 more
```

The important part is:  
__E/art     ( 8622): dlopen("/data/app-lib/eu.intermodalics.tango_ros_streamer-1/libnodelet_manager.so", RTLD_LAZY) failed: dlopen failed: could not load library "libtango_3d_reconstruction.so" needed by "libnodelet_manager.so"; caused by cannot locate symbol "rand" referenced by "libtango_3d_reconstruction.so"...__

I don't have the time to look more into it, but I suspect it's because the Tango Development device is running an old version of Android. Moreover it is not supported by Tango anymore. The two devices supported by Tango are now:
Lenovo Phab 2 Pro 
Asus ZenFone AR

The best solution would be to get one of these devices.
Otherwise, if you don't need the functionality of Tango 3D Reconstruction, you can try to remove the code where tango_3d_reconstruction is used and build the app from source.

Hope this helps.

dan9thsense, 2018-01-27:
Thanks for looking into it.  I hoped that I finally found a use for that Tango device, as it never lived up to the hype.

jubeira, 2018-01-29:
@dan9thsense have you tried an older APK version? If I remember correctly older versions didn't have those features, so they may work with the Tango development kit, and you won't need to build the app from source.
Go to https://github.com/Intermodalics/tango_ros/releases and try one of the older APKs; install them using ADB. to your Tango device.

dan9thsense, 2018-01-29:
That's a good idea.  Do you have a suggestion for which older version to try first?

jubeira, 2018-01-31:
The second release could be a good starting point: https://github.com/Intermodalics/tango_ros/releases/tag/v1.0-15. I remember using it with Tango development kit. If you need more features you can try some newer versions.

dan9thsense, 2018-02-02:
That release almost works.  It comes up with both the ROS and Tango lights green, but then the Tango light turns red.
The log says:
Successfully connected to Tango
Then the next two lines are:
tango_ros_node:388 Error, could not get a first valid Pose
tango_ros_util.cpp:62 Could not connect to the Tnago service.

PerrineAguiar, 2018-02-07:
@dan9thsense I tested the old release pointed by @jubeira on a Tango Development device and it works fine.

Can you try again and be sure to point the camera of the device to a "feature rich environment" (e.g. pointing at your surroundings, not lying on a table). That could be an issue, since the app needs a first _valid_ pose.

Also, in this version of the app, to receive data on topics, you should enable each type of data you want from the navigation drawer (press the menu icon on the right corner).

dan9thsense, 2018-02-07:
Thanks for investigating this.  I will try that on Monday, as I am traveling this week.  The camera was just pointing at a wall, so perhaps that is the issue.

dan9thsense, 2018-02-16:
Good news!  That fixed it-- data and images are coming through.

Thanks for your time and persistence to address this issue.  You have resurrected my Tango device and, for the first time, it will be a useful device.

closed 2018-02-16 by dan9thsense

tim-fan, 2018-06-03:
Hello, thanks for the above info - I have got my hands on a spare Tango development device (Yellowstone tablet, same android/kernel/tango-core version as dan9thsense's), and with the help of this thread I have been able to get it publishing to ROS. Thanks!

Worth noting - I have tried some more recent releases of the streamer app, and found that all versions work from 1.0-15 up to and including 1.3.0.  It looks like it was the 1.3.1 changes which broke the app for the yellowstone tablet. 

This makes sense, as the release notes mention changes for supporting tango core m20, while the last tango core release that supported the yellowstone tablet was Hopak.

So I'd recommend future people trying to run tango_ros_streamer on a yellowstone tablet stick to release 1.3.0 (https://github.com/Intermodalics/tango_ros/releases/tag/v1.3.0)