Intermodalics/tango_ros #385: Empty mesh extracted

MetaI, 2018-02-08:
Hi. First things first: thank you for this wonderful app, It is being very helpful for me in in order to accomplish 3d mapping tasks mounted in a small vehicle. The question is I would like to add some feedback while recording data and for that purpose I would like to enable 3d mesh reconstruction in rviz. But I'm finding unable to show the mesh markers. 
I got two devices: Zenfone AR and Lenovo Phab 2 Pro, and in them both and updated to last Tango version.
I'm loading rviz config and I can see everything except for mesh (example with Zenfone AR):

[image]

I checked rostopics and from time to time I receive data from /tango/reconstruction/mesh and /tango/reconstruction/mesh_marker

Later I wen to adb debug for checking the device output while scanning, and I got this:

```
 02-08 12:17:29.555 22784-23395/? I/native: tango_3d_reconstruction_helper.cpp:180 Empty mesh extracted.
02-08 12:17:29.930 22784-23395/? I/native: tang
![captura de pantalla de 2018-02-08 12-52-24](https://user-images.githubusercontent.com/597717/35972506-6d7259d0-0cd2-11e8-9c3b-41cb94c26c47.png)
o_ros_node.cpp:1150 Empty mesh array!
```

Any clue what I'm doing wrong? I checked ros parameters and seem to be fine. Also I use usb thetering for avoiding wifi delay. Enable depth camera and color camera are on (except for Lenovo Phab 2 Pro that color camera can't be enabled because Android version)

PerrineAguiar, 2018-02-14:
Hi @MetaI, thanks for the kind words.

I could not reproduce your issue with a Lenovo Phab 2 Pro.
You can double check that the related rosparams are set correctly:

```
$ rosparam get /tango/enable_depth 
true
$ rosparam get /tango/enable_color_camera 
true
$ rosparam get /tango/enable_3dr_mesh 
true
```

One side note, on the Lenovo Phab 2 Pro the color camera is enabled by default, and can not be disable due to Android version (I agree it is not very clear in the app). So you should receive a color image. 
Then if depth and 3dr_mesh are also enabled, you should received meshes on `/tango/reconstruction/mesh`.

According to the device output, the params are set correctly but the meshes extracted by Tango are all empty (see [here](https://github.com/Intermodalics/tango_ros/blob/0c778888ca97279b8842fcad8c69864489574e12/tango_ros_common/tango_ros_native/src/tango_3d_reconstruction_helper.cpp#L180)), so in the end the marker array is empty (see [here](https://github.com/Intermodalics/tango_ros/blob/master/tango_ros_common/tango_ros_native/src/tango_ros_nodelet.cpp#L1150)). I can't really understand why. It usually happens that some extracted meshes are empty, but not all.

Some links to the Tango3DReconstruction API that could be useful while investing this:

* [Tango3DR_extractMeshSegment](https://developers.google.com/tango/apis/c/reconstruction/reference/group/reconstruction#tango3dr_extractmeshsegment)
* [Tango3DR_Mesh](https://developers.google.com/tango/apis/c/reconstruction/reference/struct/tango3-d-r-mesh#struct_tango3_d_r___mesh)

MetaI, 2018-02-14:
Hi  @PerrineAguiar;
Thank you very much for your response! Indeed this world of indoor positioning and robotics is starting to engage me a lot thanks to your app.
Could we try to discard software versions firsts?
I'm using, in Lenovo Phab 2 Pro:
- Tango Ros Streamer from Play Store v1.3.1
- Ubuntu 16.04
- Ros lunar (also tried ros kinetic with same results) from 
`sudo apt-get install ros-lunar-desktop-full`
- Tango core m20
- Server parameters are true, the three of them

Anything different?

PerrineAguiar, 2018-02-21:
Here are what differs for me:
* Ubuntu 14.04
* ROS Indigo
* Rviz version 1.11.18

MetaI, 2018-03-02:
Dear @PerrineAguiar;

I tried Indigo in Ubuntu 14.04 and everything is working! I have to do more test, but it seems than in kinetic its not working, maybe due to ros message differences?
Thank you for your support,

Regards

PerrineAguiar, 2018-03-05:
@MetaI Good news!
I am not aware of changes in the message definition of [MarkerArray](http://docs.ros.org/api/visualization_msgs/html/msg/MarkerArray.html). 
If you find the root cause of the problem, please add a comment to this issue.
Thanks